AndyLevesque/QuackWorks #17: Additional holding strength

chelming, 2025-01-15:
https://github.com/AndyLevesque/QuackWorks/blob/35e6f8b3ab1872900fb7485a84258255beccd424/Underware/Underware_I_Channel.scad#L95

I don't notice any difference in the rendered model when modifying this line. Is there something else I need to change? I've tried 0.4 - 1.5 and don't see any difference in the profile. Should I?

AndyLevesque, 2025-01-24:
The additional holding strength right now only applies to the original profile. Were you trying it on the 2.5 profile?

chelming, 2025-01-24:
good question! I downloaded the channel scad file from this repo and tried changing the value. how do I know which version it is? 😅

chelming, 2025-01-24:
o shoot. I see the original vs 2.5 at the top now 🙃

chelming, 2025-01-27:
is there any info on the original vs v2.5? is original what's on makerlabs?

does 2.5 have additional holding strength already compared to original?

AndyLevesque, 2025-04-09:
@chelming - Both options have been released to MW. The 2.5 is an inside-out profile (which does not scale with the holding strength modifier). The original profile does scale. Folks have found the most success with the original profile and using a holding strength modification of 0.3 to 0.6 or so. The 2.5 profile is more for future plans (and therefore I might remove for now until those other features are relsased.

closed 2025-04-09 by AndyLevesque