AndyLevesque/QuackWorks #31: [Enhancement] Front clips

chelming, 2025-03-24:
being able to generate front clips like this would be awesome.

https://www.printables.com/model/221920-caldigit-ts4-bracket

AndyLevesque, 2025-04-09:
I started this work, but I need to sort out the challenge of print orientation (upright). If you have suggestions or ideas on how to work around this difference, it would help a lot!

chelming, 2025-04-09:
I merged those with a top-only holder and printed it upright, clips at the top. worked fine. [image]

AndyLevesque, 2025-05-02:
How well did the overhang print? that's what I'm trying to avoid. If it didn't have an issue, then this would be easy to incorporate

chelming, 2025-05-02:
totally fine.  the only issue is that the layer lines aren't going the correct way for strength. I snapped one off yesterday.

AndyLevesque, 2025-07-18:
Thoughts: 

The clips should be printed separately and on their side for layer strength and avoiding overhangs. The question becomes the connection method that works with the 3mm wall and where the male and female can be printed in different orientations. 

It's possible that the [rabbit_clip() from the BOSL2](https://github.com/BelfrySCAD/BOSL2/wiki/joiners.scad) library could work here with both parts having females and separately printing a male clip. The item holder would have no bridging and the clips themselves would have limited bridging and no overhangs.