AndyLevesque/QuackWorks #37: [Feature request] openGrid lite mount for underware

FaustXVI, 2025-04-23:
Thanks to David D, we now have openGrid and [an underware option](https://makerworld.com/en/models/1175037-underware-for-opengrid#profileId-1195039) that overall consume less filament.

I saw that you already generate openGrids (and the stack option is awesome!).

However, the snapping-one-side-only channel solution of David D is only available for single channel and fixed size.

It would be awesome to have the current flexibility of generators while using that new solution.

The good news is, I am a developer and would be happy to do a PR. The bad one is, I am totally new to CAD in general, even more to openSCAD and would have no idea of where to start.

pleite, 2025-04-23:
Hi! Glad you enjoy the new openGrid from David D. We have actually concentrated most of the openGrid, including, the still beta generators, on this repository: [https://github.com/[openGrid-3D/openGrid-openSCAD](https://github.com/openGrid-3D/openGrid-openSCAD)

I have also published, for convenience for those that don't want to deal with GitHub, on Makerworld: [https://makerworld.com/en/models/1329404-beta-underware-for-opengrid#profileId-1367368](https://makerworld.com/en/models/1329404-beta-underware-for-opengrid#profileId-1367368
)

FaustXVI, 2025-04-23:
That's totally the answer I was looking for :) Thanks a lot !

closed 2025-04-23 by FaustXVI