AndyLevesque/QuackWorks #38: [Feature request] Hooks

FaustXVI, 2025-04-23:
The first version of underware offered hooks that are nice to roll long cables.

Could we have a generator for them ? (They would then become trivial to adapt to openGrid)

Again, I'd be happy to do a PR but am totally new to openSCAD (meaning, changes will take me some time and heavy review would be necessary).

FaustXVI, 2025-04-23:
If anyone coming here has the same needs, [I just found an option](https://www.printables.com/model/1264943-underware-parameteric-cable-spool-hook-for-opengri)

pleite, 2025-04-23:
Hi! The [Underware_Cable_Loop_holder.scad](https://github.com/AndyLevesque/QuackWorks/blob/main/Underware/Underware_Cable_Loop_holder.scad) is the one on this repository. If you change the distanceBetweenSlots in the customizer section to 28 you become openGrid compatible.
There were former cable hooks with a different look, like multiple hooks and a cover, these are not yet here as the static versions were sufficient. Can you clarify if you intended this one, or the later?

FaustXVI, 2025-04-24:
Hi !
I was talking about the former cable hooks indeed.
The version you linked to would do the job for me but : 
 - It doesn't appear on maker world generator. Is it on purpose ?
 - It uses more filament than the former hooks.

The shapes **seems** simple enough to me to feel able to attempt to create a generator for it. I'll try to make a PR for it.

pleite, 2025-04-24:
Ping us on discord if you need help, should be quite doable. The old one was screw only but I would do it as multiconnect using the backplate you have, start with this one for example if you’d like.

PR #42 (2025-05-02): First draft of hooks
Here is a first draft of hooks.
I shamelessly copied the Item holder and adapted from their.
That the first thing I ever try to do with openscad so please give me feedback :) I have much to learn !
Fixes #38
closed 2025-05-02 by AndyLevesque