AndyLevesque/QuackWorks #43: opengrid scad - Add spaced option for mounting holes

codyolsen, 2025-04-26:
I'm working on an option to add spacing across the grid. Not quite polished yet, but posting that it's active here in case anyone is working on a similar task in paralell.

https://makerworld.com/en/models/1304337-opengrid-tile-generator

[image]

pleite, 2025-04-27:
Hi! Would you like to submit a PR to this repository too?
I have placed a PR that we are reviewing to add threads of know screws and would fit nicely with those.

codyolsen, 2025-05-05:
Yeah when it's ready I'll push up a PR. I'm still prototyping the spacing modulus to make sure it makes sense in the generator UX. Not totally confident in the first version so don't wait for the screw size pr for me. :D

AndyLevesque, 2025-07-18:
PR accepted and merged

closed 2025-07-18 by AndyLevesque