AndyLevesque/QuackWorks #64: Round holder - Fix slot cutout

AndyLevesque, 2025-07-18:
Comment from MakerWorld (https://makerworld.com/en/models/582260-multiconnect-part-generator-master-collection#profileId-633542)

[@Typ_im_dunkelln](https://makerworld.com/en/@Typ_im_dunkelln/upload)
[@BlackjackDuck](https://makerworld.com/@BlackjackDuck) FYI: The generator for the round holder has a bug, when using `itemDiameter` 83mm, `rimHeight` 40mm and then add a slottet cutout (I used `holeCutoutDiameter` 50mm). With those values the slot does only cover half the height, leading to open, bend bridging of `holeCutoutDiameter` mm.

I fixed it by doubling the height of the cutting cube in code.
```
                    if (slotCutout == true) {
                        translate(v = [-holeCutoutDiameter/2,0,0])
                            cube([holeCutoutDiameter, totalWidth/2, totalHeight*2]);
                    }
```

mmodrow, 2025-07-18:
I just noticed, that the ad hoc solution is not touching the root cause:

totalHeight = max(baseThickness+shelfSupportHeight+0.25*itemDiameter,25);

This is used as the cutout height, which is understandable. Yet, it does not take into account the rimHeight at all, yet it takes into account the itemDiameter. That seems odd.

timtucker, 2025-07-20:
Linked some changes to the code that tweak things so that the hole / slot be repositioned, but can't extend backwards into the slots.

With the updated logic & configurator you can have slot start and end anywhere along the Y axis you want (which means support for things like putting a cord cutout at the back for an Amazon Echo)

PR #80 (2025-07-20): Rework rounded item generator
Got started playing around with OpenSCAD for the first time and went a little overboard:
- Curved support structure (should be stronger, print faster, and use less filament)
- Allow oval objects
- Allow generation of rectangular items
- Allow the hole / slot to extend arbitrarily forward & backward - should address #27 
- Prevent the hole / slot from extending into the back wall - fixes #64 
- Allow tuning for the number of slots wide / high
- Allow tuning of the resolution of circles and curves
- Simplify / reorganize parameters

This tackles at least a few of th [... cut]
closed 2025-07-20 by AndyLevesque