AndyLevesque/QuackWorks #94: (Bug) [Deskware] Deskware_Main.scad Doesn't Generate Top Plate for End Plates

quagzlor, 2025-08-16:
Was assembling my parts and something felt off. Checked and realised that the Top Plate that goes on the end risers isn't being generated in Deskware_Main.scad, when downloaded from Github.
Shows up fine on the Makerworld file.

[image]
[image]

AndyLevesque, 2025-08-16:
Can you share the parameters you are using to get this behavior? I'm not able to recreate on my end.

quagzlor, 2025-08-16:
I was able to figure out the issue!
You have to enable the 'roof' module since it isn't enabled by default.
My bad, i'm new to OpenSCAD and was following the Installation Instructions